RobTillaart/DHTNew #5: Example needed for waitForRead flag

RobTillaart, 2020-05-01:
version 0.1.7 supports a waitForRead flag, hoever an example showing its use is missing at the moment. The example should show the use and the added value of the flag. If needed a readme.md file to explain the example should be added.

RobTillaart, 2020-05-03:
@Mr-HaleYa 
As you asked for this waitForRead functionality could you mind writing a small example sketch showing the effects of using this flag or not.? 
Please keep it in line with the other examples

Mr-HaleYa, 2020-05-03:
Yes when I have some free time off of my current project I will write an example sketch. I live on the west coast of the USA so it is very early in the morning when you get online 🙃

Mr-HaleYa, 2020-05-03:
I modified the test sketch so it stayed inline. I rearranged the columns and added an AGE column to the start to tell if it is a NEW or OLD reading. It works exactly as it is supposed to. I kept the sensor TYPE column so if they were to change the sensor type while testing they would see the change in the sketch and the time difference in the TIME column for when WaitForReading is set to true.

RobTillaart, 2020-05-04:
Thanks, also for catching the typos.

closed 2020-05-04 by RobTillaart