RobTillaart/DHTNew #9: Not update on platefomeio

fmatray, 2020-05-21:
Hello,
The version on PlateformeIO is 0.1.5 and not the last one.
I have a workaround.

Can you, please, update it ?
Thanks.

RobTillaart, 2020-05-21:
I am updating libraries and moving them to their own repo. This makes handling issues easier.

Will check on this asap

RobTillaart, 2020-05-22:
PLatform IO probably takes the copy from my combined repo https://github.com/RobTillaart/Arduino

I updated the DHTNEW lib there too, so now wait until it is picked up by platformIO

Feel free to (re)register the lib yourself, I never registered any of my libs @ platformIO, 

 https://docs.platformio.org/en/latest/core/userguide/lib/cmd_register.html#cmd-lib-register

fmatray, 2020-05-24:
Hello,
thanks a lot but I still have a problem. Only the library.json is downloaded, I tried several time. 

I also tried to register the library on PlatformIO but it looks like some informations are missing in library.json.

user@prometheus DHTNEW_ID5548 % platformio lib register https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTNEW/library.json
Error: Could not parse manifest -> Expecting value: line 6 column 1 (char 5)

user@prometheus DHTNEW_ID5548 % platformio lib register https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTNEW/library.properties
Error: Invalid manifest fields: {'description': ['Length must be between 1 and 1000.'], 'version': ['Missing data for required field.']}. 
Please check specification -> https://docs.platformio.org/page/librarymanager/config.html

user@prometheus DHTNEW_ID5548 % platformio lib register https://github.com/RobTillaart/DHTNew/blob/master/library.json                  Error: Could not parse manifest -> Expecting value: line 6 column 1 (char 5)

RobTillaart, 2020-05-24:
Seems like I made an syntax error in the json file. Need to check the specification. 
Thanks for the error msg and reporting

RobTillaart, 2020-05-24:
First I look at the second part of the error message:

**description field**
verified against - https://docs.platformio.org/en/latest/librarymanager/config.html#description
The description field is a string,
the length is smaller than 255 
the length is 95 so between 1 and 1000 (the error message)
So it seems to me that the error message does not mention the real/correct cause

**version field**
https://docs.platformio.org/en/latest/librarymanager/config.html#version
The version field is a string
The length is smaller than 20
The format is according the semantic versioning rules, three numbers separated by dots .

fmatray, 2020-05-24:
I agree and I don't understand the message.

I compared with another projet and I can't find any problem.

RobTillaart, 2020-05-24:
Same bug is mentioned - https://github.com/platformio/platformio-libmirror/issues/173
but in that case the version field was indeed not available

RobTillaart, 2020-05-24:
installed platformio on my computer - c:\users\Rob\.platformio\penv\Lib\site-packages

in that folder there are the following python scripts:
**platformio\package\exception.py**  holds the Invalid fields text message [line 34]
printed after a ManifestValidationError

**platformio\package\manifest\schema.py** raises this ManifestValidationError [line 41, 53]
these lines are probably used by ManifestSchema(baseSchema)   [line 150]

this is called in **platformio\commands\lib.py**  [line 505]
that calls  ManifestParserFactory.new_from_url(config_url)  which is in **parser.py**

and that loads the file.

RobTillaart, 2020-05-24:
No clue from the code..

The other error

platformio.exe lib register https://github.com/RobTillaart/Arduino/tree/master/libraries/DHTNEW/library.json 
returns  **Error: Could not parse manifest -> Expecting value: line 6 column 1 (char 5)** 

platformio.exe lib register https://github.com/RobTillaart/DHTNew/library.json
just crashes in a long stacktrace resulting in a 404 error

platformio.exe lib register https://github.com/RobTillaart/DHTNew/blob/master/library.json
Error: Could not parse manifest -> Expecting value: line 6 column 1 (char 5)

So at least after a few hours I am able to reproduce the other error

RobTillaart, 2020-05-24:
DId a rewrite of the library.json file - https://github.com/RobTillaart/DHTNew/pull/10
but still got the same error ..

RobTillaart, 2020-05-24:
OK, I noticed now that you did use the **library.PROPERTIES** file in the second error message while platformio expects you to give **library.JSON** file.  

WRONG parameter ==> So that one is solved

fmatray, 2020-05-24:
I tried both, JSON and PROPERTIES, and I get different errors.

As mentioned in URL below, register can take library.json, library.properties or module.json. Only one is needed to register.
https://docs.platformio.org/en/latest/core/userguide/lib/cmd_register.html#description

```
% platformio lib register https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTNEW/library.json
Error: Could not parse manifest -> Expecting value: line 6 column 1 (char 5)
```

RobTillaart, 2020-05-24:
Im new to platformIO too. But the error message given on the properties file are sbout missing json fields. So the tools used are not that robust yet. 

The other error is still strange, will investigate coming days.

RobTillaart, 2020-05-25:
> I tried both, JSON and PROPERTIES, and I get different errors.
>
> As mentioned in URL below, register can take library.json, library.properties or module.json. Only one is needed to register. [...]
As far as I understood the register code, the error message came from parsing the library properties file as if it was a library.json file.

The "line 6" error is still there, as said 3 post back, will be working on it later when time permits.

RobTillaart, 2020-05-25:
Had a quick check with two repo's that have a different library.json file than mine
``` 
platformio.exe lib register https://github.com/PaulStoffregen/OneWire.git
platformio.exe lib register https://github.com/milesburton/Arduino-Temperature-Control-Library.git
```
both gave the same "Line 6" error message on a completely other repository, other file.

So I think you should file an issue at the platformio.org site as the library.json file follows the documentation and the error is seen on at least 3 different instances of the file.

Please let me know if you get a solution from platformio people.

alrabe, 2020-05-31:
You get the library working when referring directly to it in your projects platform.ini file:

`lib_deps =
  https://github.com/RobTillaart/DHTNew`

RobTillaart, 2020-06-02:
@fmatray   
Can you confirm the solutionof alrabe?

RobTillaart, 2020-06-16:
As I got no feedback I assume the problem is solved.

closed 2020-06-16 by RobTillaart