RobTillaart/DHTNew #25: Hey Rob,

Mr-HaleYa, 2020-06-18:


RobTillaart, 2020-06-18:
HI, long time no see ,

Yeah, played a lot with different ideas to make things easier for the users. Storing the values internally as an integer is one of them. It saves 2 bytes RAM for humidity , 2 bytes RAM for temperature and 2x3 bytes for the offsets. It adds four multiplications to be able to return floats, but it would remove some float math in the read function so that compensates. 
Eight boolean flags can be stuffed into one byte so that could save a few bytes but the lib only has one boolean flag left.  So in the end 10 bytes of RAM would be saved. Did not spend time implementing it yet.

> My temp is borderline 23-24°C so it bounces between these two like crazy on my graphs so I was checking to see if decimals where an option as a reply from the sensors.

What you could do is add a low pass filter. I give an example in simplified code
```cpp
float lowPassTemp;
const float alphs = 1.0 / 3;  // any value between 0.001 and 0.999 will do.

setup()
{
  lowPassTemp = sensor.getTemperature();
}

loop()
{
  float currentTemp = sensor.getTemperature();
  lowPassTemp =  lowPassTemp + (currentTemp - lowPassTemp ) * alpha;  // << the trick
  Serial.print(currentTemp, 2);
  Serial.print("\t");
  Serial.println(lowPassTemp, 2);
}
```

The idea is that you do not replace the old temperature with the new measurement but make a step in between those two values. How much in between depends on the value of **alpha**.  If it has the value 0 the temperature will not change at all, if it has the value 1 it will replace the old temperature completely.  Another way to write it is more clear for user (uses more math)
```cpp
lowPassTemp =  (1 - alpha) * lowPassTemp + alpha * currentTemp;
```

Please note that measurements will be smoother but the system will also react a bit slower to variations. You need to check what value **alpha** feels acceptable for you / your customer.

In a project I made it user configurable.
``` cpp
alpha = 0.01 + analogRead(A0) / 1024.0;
```
so the customer could tune it with a potentiometer. 

In the above alpha is a constant but you can make alpha a variable that depends on other parameters. E.g. if delta T is large make alpha also larger to follow large changes faster but stay smooth when delta T is small. Or you can make alpha depends on light conditions or UV radiation, you name it. Anything can make sense. 

Enough said, please give it a try.

RobTillaart, 2020-06-18:
Added the integer storage as issue #26 to investigate

Mr-HaleYa, 2020-06-18:
hahahahahahahaha oh my gosh, Rob how do you even have the patience to write these out? 🤣 
I was just wanting replies from ready(); in float so I could do decimals lol. 

I don't know if I explained already the DHT sensors as internal temperature and humidity sensor for my project, I don't even know if we need it to go out to a decimal point because all it does is make sure that they are not overheating because they will be mounted on poles in the direct sun all summer and we need to make sure that the sun is not cooking our components or our batteries because we don't want explosions LOL. I guess this was just more of a thing that was bugging me then it is an actual problem because it's my house temperature that is fluctuating between 23 and 24 but outside it will be way bigger fluctuations in that so it will not matter on my graph...

I do think that there should be an option for float because people buy the sensors such as the dht22 for accuracy and if you're eliminating an entire integer of accuracy (assuming that it is 0.9 and it rounds down naturally, and they're not using the thing you posted above) then that might frustrate some people since they are not able to "fully utilize the hardware that they purchased because the library does not allow it". Even if it does take up a little more space I think there will be some people that appreciate the capability of being able to have the decimal place, especially if someone is trying to do very tight temperature control or humidity control. Whether to have it as a flag that is optional or not is a different story, but I guess it's totally up to your decision as is everything in this Library LOL

Mr-HaleYa, 2020-06-18:
Oh, I understand what your example does. It basically artificially makes the float after receiving the reading. (just so you know you have a typo that says alphs instead of alpha where you declare it) 

Just as a side question, I have never asked this of a GitHub maker but are these libraries free to use if you were to build a product and sell it to people or am I going to have a 6 ft 7 Rob kicking down my front door demand a cut of the profits? I am genuinely curious as I am very close to finishing these and once they are finished they will be available for sale and I don't want to get in any kind of trouble lol

RobTillaart, 2020-06-18:
NEVER NEVER NEVER place the sensors in direct sun light.
- it will give a systematic too high reading for temperature 
(think humidity might be lower, not sure on that one)
- it might damage in the end

Outside temperature must always be measured in a well ventilated shade
-> https://www.youtube.com/watch?v=3uBv7A7aqXE  
might be overkill but you get the idea

Mr-HaleYa, 2020-06-18:
No, you got it all wrong lol they will be inside of a box and the Box will be in the Sun. I'm hoping not all the time but I may not have control over that's why I want to use them to test how hot it gets inside the box so I need to know how bad I need to cover it with shade or ventilation 

Yeah I discovered not to put them in the direct sun I put a dht11 in direct sunlight and it gave me 128 degrees °F which isn't even in the range of the sensor

RobTillaart, 2020-06-18:
With the integer trick I store not whole degrees but tenth of degrees.

e.g. when temp is 31.4 degrees I will store 314 in an integer and multiply with 0.1
As the temperature goes from -40 .. +80 I just make that -400 .. + 800 
HUmidity from 0 - 100.0% ==> 0 .. 1000

RobTillaart, 2020-06-18:
> Just as a side question, I have never asked this of a GitHub maker but are these libraries free to use if you were to build a product and sell it to people or am I going to have a 6 ft 7 Rob kicking [...]

Depends on the license attached, when you sell it you must follow those.
Mine are MIT licensed, which means you are free to use it with no warranty or liabilities.
You may not sell it as your own work and you must provide reference to the original
copyright and license note.

Github shows summary if you click on the license file.

Mr-HaleYa, 2020-06-18:
He is an image of the box it will be stored in, It is fully waterproof (if that's even a real thing) so there is no airflow... So I am testing with a dht22 to see how hot it gets and how different things affect it such as shade/vent holes (this shows a DHT11 top leftish but it has since been changed out for a dht22 after 3.0 release 🙂 )

[image]

RobTillaart, 2020-06-18:
I do not know what other libraries you use, but legally you must check them all.
That is why I write my own most of the time.

But hey if I saved you a lot of money, you can always donate :)

Mr-HaleYa, 2020-06-18:
> But hey if I saved you a lot of money, you can always donate :)

haha, I was planning on it once i got some revenue flowing, for all the help you have been. (Remember I asked about buying you coffee lol)

RobTillaart, 2020-06-18:
> He is an image of the box it will be stored in, It is fully waterproof (if that's even a real thing) so there is no airflow... So I am testing with a dht22 to see how hot it gets and how different things affect it such as shade/vent holes (this shows a DHT11 top leftish but it has since been changed out for a dht22 after 3.0 release 🙂 ) also most of the hardware is different now...
> 
> [image]

Looks good, IP67 watertight ? 
That means it will not measure outside humidity, just the one in the box.
A tricky thing might be that if water leaks in (after service or whatever) it will never ever get out.

RobTillaart, 2020-06-18:
What does the project do? (no need to answer if you do not want to)

Mr-HaleYa, 2020-06-18:
Essentially, distance sensing. Yes, I understand that it won't measure the humidity outside the box, that's why I put it inside the Box I just need to make sure that the inside of the box stays dry enough that the electronics don't short out and cooling enough that they don't melt LOL. It can stay charged for about 8 days without solar power and charge within one day so I feel that the solar efficiency is decent.

Basically the DHT just functions as a troubleshooter. I can look at the history graph and if I see a 100% spike in humidity right before it died it probably got wet or a 600-degree spike in temperature it probably vaporized

RobTillaart, 2020-06-18:
> Yeah I discovered not to put them in the direct sun I put a dht11 in direct sunlight and it gave me 128 degrees °F which isn't even in the range of the sensor

Nice test, never did that ...

The box is gray so it will not get that hot but you could spray it with a waterproof white paint to make it even cooler.

Mr-HaleYa, 2020-06-18:
Built this beauty last night and I'm planning on my first full test outdoors with all the new hardware within the next 2-3 days. I've done plenty of outdoor tests before but never with all this new hardware. I measure my External temp with a DS18B20 probe that has a solar shield (the white thing you can barely see in the bottom, black in the one above)

[image]

Mr-HaleYa, 2020-06-26:
Well, I'm happy to report that mine has been running for 7 days with the dht22 and esp32 and not an error wrong reading! Fantastic work with that fix 👍

RobTillaart, 2020-06-27:
Good to hear! I did not encounter other problems too. 

* Is it fixed? 
Suppose so, the workaround behaves very robust as in 7 days we would expect ~14 crashes and none happened. So simpified math says chance = 1 in  2^14 (it happens or not) = 1 in 16000 so 3 more days to come at the level of 1 in a million.
* Is it understood? 
Definitely not 100%, but we are pretty close to the core issue. it is a timing problem during wake up of the sensor in combination with ESP32 happening at predictable intervals.
* Is further investigation needed? 
Yes, I would like to (1) measure the deviation in the timing  (2) understand if the root cause comes from the sensor or in the ESP32.  This 2nd point could prevent similar issues with other time critical sensors in the future. But for now my priorities are on refactoring my other libraries and open issues.

Thanks again for the feedback!

PS,
Can we close this issue?

Mr-HaleYa, 2020-06-27:
I will let you know if any problems arise, thank you for the work you did in fixing it. I will close this now since you have provided a solution to my OG question.

closed 2020-06-27 by Mr-HaleYa