RobTillaart/DHTNew #46: pin setting at runtime

app-vrb, 2021-01-04:
Hi,
I am looking for a library with the possibility to define the pin at runtime.
It is possibile with this one?
Thanks

RobTillaart, 2021-01-04:
You can create runtime an DHTNEW object with the pin you want.
Never tried it however, pleas elet me know if it works and I will add it to the examples

```cpp
//
//    FILE: dhtnew_runtime.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.0
// PURPOSE: DHTNEW library test sketch
//     URL: https://github.com/RobTillaart/DHTNew
//
// HISTORY:
// 0.1.0    2020-01-04  intial version
//
// DHT PIN layout from left to right
// =================================
// FRONT : DESCRIPTION
// pin 1 : VCC
// pin 2 : DATA
// pin 3 : Not Connected
// pin 4 : GND

#include <dhtnew.h>

DHTNEW mySensor(16);

uint32_t lastTime = 0;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.println(DHTNEW_LIB_VERSION);
}

void loop()
{
  if (millis() - lastTime > 2000)
  {
    for (int pin = 6; pin < 10; pin++)
    {
      DHTNEW sensor(pin);
      sensor.read();
      Serial.print(pin);
      Serial.print("\t");
      Serial.print(sensor.getHumidity(), 1);
      Serial.print("\t");
      Serial.println(sensor.getTemperature(), 1);
    }
  }

  // Do other things here
}

// END OF FILE
```

app-vrb, 2021-01-04:
Hi Rob thanks for your answer.

One question:
The line:
> DHTNEW mySensor(16);
It is  mandatory to be before the setup()?
The question is because I know the pin just inside the setup().

Thanks again for your support.
Francesco

RobTillaart, 2021-01-04:
ooops , copy paste error  ... 
I will remove it from the example

commit b7105e9a (2021-01-12): fix #46 - new example
closed 2021-01-12 by RobTillaart