RobTillaart/DHTNew #51: error yield() no se declaró

germe-fur, 2021-01-20:
(spanish)
### i tried to use the library and i get this error:
the code:
```C
#include <dhtnew.h>

void setup(){
  
}
void loop(){
  
}
```

compiler:
```/home/germe/Documentos/arduino/libraries/DHTNEW042/dhtnew.cpp: En la función miembro ‘int DHTNEW::read()’:
/home/germe/Documentos/arduino/libraries/DHTNEW042/dhtnew.cpp:112:13: error: ‘yield’ no se declaró en este ámbito
       yield();
             ^
/home/germe/Documentos/arduino/libraries/DHTNEW042/dhtnew.cpp: En la función miembro ‘int DHTNEW::_readSensor()’:
/home/germe/Documentos/arduino/libraries/DHTNEW042/dhtnew.cpp:248:9: error: ‘yield’ no se declaró en este ámbito
   yield();
         ^
```
the compiler is in spanish because my system is in spanish. it say `"error in the member function ‘int DHTNEW::_readSensor()’:`
`yield not declared in this ambit`
sorry for my english, i speak spanish

my library version is:`0.4.2`
i'm on arduino ide `2:1.0.5+dfsg2-4.1`
the error ocurrs when i import the library. if i put more code the error splashes too

RobTillaart, 2021-01-21:
Thanks for using my library and give feedback,

```yield()``` is a function that is used by many operating systems to indicate that now it is a good time to swap processes.  On embedded systems like ESP32 it is used to handle background processing like WIFI handling etc that are time critical.

On a standard Arduino UNO the function is implemented as an empty function, that will be optimized away by the compiler. 
The more recent versions of the "Arduino compilers" should support this function.

Several options:
- The easiest way to handle it is to remove the line ```yield();```, 
- upgrade your compiler, but that may still not support it => file an issue there
- add an empty function to your Arduino.h file  - as you will encounter more libraries / code that calls ```yield()```

```cpp
void yield() {};
```

I cannot remove it from the library as it is there for a reason as you understand.

Hope this helps,

RobTillaart, 2021-01-21:
If you have time to play ;)

https://github.com/RobTillaart/infiniteAverage

germe-fur, 2021-01-22:
thanks for the response!!
it now work!
(i added an empty void yield)

closed 2021-01-22 by germe-fur