RobTillaart/DHTNew #57: Still problems with measuring negative values (AM2301)

matchacos, 2021-04-08:
Hello
I am using several senses (dht22, am2301) to manage domotics at home and my outside sense blocked at 0°C while descending during the night.
Actually I tried another set of senses in my fridge and actually it appears that the two last versions of you library doesn't function quite well with (at least) the am2301 senses, while I installed your 0.4.3 version and had correct values.
In facts the 0.4.5 version only gives me 0°C for negative values.
The 0.4.4 version gives me -3267 +/- values when negative.
Perhaps it is linked to your last modifications like in the last issue (#52 ) ...?
Thanks for your work anyway and good luck

matchacos, 2021-04-09:
The way the AM2301 codes negative temperatures seems to be the same than the AM2302 does.
Here is the datasheet for this item:
http://www.aosong.com/userfiles/files/media/AM2301%E8%A7%84%E6%A0%BC%E4%B9%A6-%E8%8B%B1%E6%96%87%E7%89%88%EF%BC%8820210407%EF%BC%89.pdf

RobTillaart, 2021-04-09:
@matchacos 
I looked at the temperature code of 0.4.5 and it is definitely wrong for negative temperatures.
In fact it maps all negatives on zero, just like you noticed. 

A new develop branch (0.4.6) will be created a.s.a.p.

matchacos, 2021-04-09:
Nickel! I tried it with the new files I found and it worked well.
The negative temperatures are now collected without any hiatus? No problem appeared to me.
Thanks a lot

RobTillaart, 2021-04-09:
Thank you for finding and verifying!
Will merge branch tomorrow.

R.

commit 4ff08a3a (2021-04-10): Fix #57 negative temperatures (#58)

* Fix #57 blocking temp below zero
closed 2021-04-10 by RobTillaart