RobTillaart/DHTNew #63: Humidity values higher than 100%

afonsoc12, 2021-04-13:
Hello, 

I have recently changed to DHTnew after this thread, and I noticed that completely at random I get this very high values of humidity. Of course I can just ignore them if higher than 100, but I was wondering what the cause would be.

My setup are two ESP-01s one with DHT22, the other with a DHT11, both send values to an MQTT broker, which are captured by telegraf and stored in an InfluxDB instance.

Humidity values higher than 100%

| time                           | temperature | humidity | location    | sensor |
|--------------------------------|-------------|----------|-------------|--------|
| ----                           | ----------- | -------- | --------    | ------ |
| 2021-04-11T23:56:08.272696401Z | 8.3         | 3317.3   | Living Room | DHT22  |
| 2021-04-12T03:37:10.063376651Z | 8.4         | 3316.6   | Living Room | DHT22  |
| 2021-04-12T04:57:10.712664669Z | 8.4         | 3316.5   | Living Room | DHT22  |
| 2021-04-12T07:48:12.095298698Z | 8.6         | 3316     | Living Room | DHT22  |
| 2021-04-12T08:25:42.399491599Z | 8.7         | 3315.7   | Living Room | DHT22  |
| 2021-04-12T09:14:42.796313132Z | 8.8         | 3315.2   | Living Room | DHT22  |
| 2021-04-12T10:23:13.385068097Z | 8.9         | 3315     | Living Room | DHT22  |
| 2021-04-12T11:03:43.682027118Z | 9.1         | 3314.3   | Living Room | DHT22  |
| 2021-04-12T14:58:15.595547784Z | 9.2         | 3317.8   | Living Room | DHT22  |
| 2021-04-12T15:53:46.033386665Z | 9.1         | 3318.4   | Living Room | DHT22  |
| 2021-04-12T15:59:16.076817334Z | 9.1         | 3318.4   | Living Room | DHT22  |
| 2021-04-13T13:36:56.626400588Z | 9.3         | 3317.2   | Living Room | DHT22  |
| 2021-04-13T15:49:57.682886104Z | 9.5         | 3317.1   | Living Room | DHT22  |

The graph of the last 48 hours looks like this:
[image]

Living Room is the DHT22 and Kitchen is DHT11. They are at the same location, side-by-side, despite the names

Any thoughts on why this happens?

Thanks!

RobTillaart, 2021-04-13:
A problem could be the timing of the bits clocking in. If one reads the
data line too fast or too slow, bits may shift 1 position. Effect is that
values seem to be multiplied by two.
An humidity of 55% could suddenly become 110%. To see if this happens, you
need to reverse engineer the bit patterns and compare them. Difficulty is
that only a part can be shifted.

These bit errors can be detected by checking the return value of read.
Shifted bits would typically cause a CRC error.

Hope this helps you analyzing your data.

afonsoc12, 2021-04-13:
I assumed CRC is implemented in the lib right? I only take a reading if I get a `DHTLIB_OK` code, otherwise I discard it.

I am taking reads every 10 seconds

RobTillaart, 2021-04-14:
The CRC is checked in the library and if it fails it is indicated in the return value of **read()** . Other conditions that could fail are checked too. The first error is reported.

What action to take is up to the user of the sensor. 
E.g. a CRC error implies that four bytes do not add up to rhe fifth. That doesn't tell anything about which of those five bytes has the error. 

E.g. if the last bit of H or T flips there is a CRC error but only one tenth difference. 
If the error is in th CRC byte itself, T an H

RobTillaart, 2021-04-14:
Oops, slip of finger...

RobTillaart, 2021-04-14:
> I assumed CRC is implemented in the lib right? I only take a reading if I get a `DHTLIB_OK` code, otherwise I discard it.
> 
> I am taking reads every 10 seconds

In theory the chance of a correct CRC while there is an error is small. As the bits are added the bits are not only affecting the CRC bit value on their position, but also the bits left of them. 

In practice there are scenarios where the CRC is correct when they data is wrong.
Example:  when all bits are shifted one place right, adding them can give a correct CRC.

```
00000001   00000001  00000001  00000001  00000100    - example 
==>
00000010   00000010  00000010  00000010  0000100?
```

(this example is simplified but works for most if not all bit patterns)
If the last bit that is shifted in is a 0, the CRC will be OK , when it is a 1 it will fail  (so 50% chance).
Bits shifted left one place has similar odds. 

----

In the past I wrote code that did additional checks. These checks never made it into the library. They added extra footprint and the added value seemed small. The additional tests included thoughts like:

- as T never gets above 80°C  (datasheet)  means the bits that may be set are  10000011 11111111  (5 bits always zero)
- if T is negative (-40°C) the bits that may be set are 10000001 1111111  (6 bits zero)
- as H never gets above 100% (datasheet) means the bits that may be set are   00000011 11111111  (6 bits always zero)

The zero bits can be easily and fast checked by applying a mask, but typical the CRC should detect such bit errors.
A bit mask test would catch a part of the "shift one place" errors.  One thing I never investigated is if and how this large 
percentage of fixed zero's affect the effectiveness of the CRC code.

Other and imho better ideas included
- remember previous T and H and time in millis() of last read.
- provide delta T and H   === current - previous   
- provide gradient T and H  === delta  / (millis() - lastTime)   (think 1st order derivative)
- remember last gradient 
- detect flip of gradient  (sign gradient changes)
- detect increase of gradient (early sign of runaway heat up)  (think 2nd order derivative)
- get/set warning and alert gradient (better not in the library)

This functionality would allow to detect more anomalies in the readings, and could be interesting in their own way.

RobTillaart, 2021-04-14:
Back to your original question, the humidity all hoover around  3315  in the final representation and it definitely looks like there is a constant offset. Question is where is this offset introduced?

**Sensor did it**
3315 is  the following bit pattern  00001100 11110011   (so first byte has 2 bits set incorrectly)
If I mask out these two bits  00000000 11110011  ==> 24.3%  humidity which sounds pretty OK.
All other values are in the same range so they should be OK too.

Temperature = 8.9 ==> 00000000 01011001

CRC =  00000000 +  01011001 + 00001100 + 11110011  = 11010010
without the 2 incorrect bits
CRC =  00000000 +  01011001 + 00000000 + 11110011  = 01001100

so  CRC should have caught these,

afonsoc12, 2021-04-15:
It makes sense the scenario that you outlined, such as when all bits are shiffted. But I thought that this would be very rare to occur, let alone 12 times in 2 days.

What about some simple functionality that would give an "Overflow" error if the values are outside the sensor specs? I believe the footprint of this would be rather small and very effective. You wouldn't need to check the bit values, only make sure that the temps and humidities are within the specified thresholds.

Would this make any sense?

RobTillaart, 2021-04-15:
Hi,
Overflow (Or underflow) is relative easy to detect. I will give it a try, expect a experimental branch in an hour or less..

RobTillaart, 2021-04-15:
It will be a new return value for **read()**

DHTLIB_VALUE_OUT_OF_RANGE        -99

RobTillaart, 2021-04-15:
Check out_of_range branch, read can return these two values...

#define DHTLIB_HUMIDITY_OUT_OF_RANGE     -100
#define DHTLIB_TEMPERATURE_OUT_OF_RANGE  -101

It will not detect everything e.g. T = 25°C and H = 45% , if those values shift, they still are within range.

**Gradient**
Comparison with previous value (delta/gradient) is more effective, however users typically sample so often that the max delta is in the order of resolution steps of the sensor used. For a DHT22 you typically read so often that you see at most 0.2 between reads . 

Temperatures hardly ever drop harder than 1°C a minute

(disclaimer)
```
Spearfish holds the world record for the fastest recorded temperature change. On January 22, 1943, at about 7:30 a.m. MST, the temperature in Spearfish was −4 °F (−20 °C). The Chinook wind picked up speed rapidly, and two minutes later (7:32 a.m.) the temperature was +45 °F (7 °C).
```
27C in 2 minutes. =  1C / 4 seconds.

RobTillaart, 2021-04-15:
Some typos (too eager too publish ;)

afonsoc12, 2021-04-15:
Thank you very much! I will definitely test this soon.

> Gradient
Comparison with previous value (delta/gradient) is more effective, however users typically sample so often that the max delta is in the order of resolution steps of the sensor used. For a DHT22 you typically read so often that you see at most 0.2 between reads .

I understand your point, however I am not sure if this in practice occurs. I will gather my sensor data that I have persisted and I will make these computations to check whether there are no changes higher than 0.2. I will post the results in this issue (?)

Also, for low power applications that perhaps take readings every minute, these changes can be higher. I don't really care with outliers, because the temperatures I actually use are the mean by minute (readings every 10 seconds) and I use a 3-point moving-average filter. Something like this:

```sql
SELECT moving_average(mean("temperature"), 5) 
FROM "mqtt_consumer"
WHERE ("location" =~ /^Kitchen$/) 
AND time >= now() - 24h 
GROUP BY time(15m) fill(null)
```
*5-point moving average on 15 mins bins of data, for the last 24h*

This is InfluxQL, but you get the picture.

The graph would look like this:
[image]

So these out-of-delta values are not really a problem (but the overflow ones where it reads 3200 %H were definitely one, of course)

afonsoc12, 2021-04-15:
I had that case on the Arduino code, before you implemented this on the library. Therefore, there was no use in including in the InfluxQL query again

RobTillaart, 2021-04-25:
Would it add value to merge this branch - https://github.com/RobTillaart/DHTNew/tree/out_of_range

afonsoc12, 2021-04-25:
Hi,

I have been using an external code block to validate if the value is within the ranges. I believe it doesnt hurt to include it, but thats up to you

RobTillaart, 2021-06-13:
@afonsoc12 
Been a while and gave it some time to think it over and I decided to add it behind a **#define** flag. 
This flag is default not set.

Arguments: 
- it has definitely a functional value (see discussion above)
- occurs quite seldom so most people will not need it
- code adds 84 bytes (UNO due to float comparisons)

commit b2bee0e8 (2021-06-13): fix #63 out of range flag
commit 87bff13f (2021-06-13): Add flag for out of range check (#65)

* fix #63 out of range flag
closed 2021-06-13 by RobTillaart