RobTillaart/DHTNew #64: F_CPU not defined for Arduino Nano RP2040 Connect

RudolfAtHome, 2021-06-03:
Therefore there is an error when compiling.
The authors of the RP2040 core try to justify the lack. It's just ridiculous!

RobTillaart, 2021-06-03:
Hi,

nice to hear the library got tested on the RP2040 too. No time spend with that board .

What could be added to the library is something like 
```cpp
#if !defined(F_CPU)
#ifdef ARDUINO_ARCH_RP2040
#define F_CPU 100000000     // ???
#else 
#error "need to define F_CPU"
#endif       
```

However I think that should be solved in the "Arduino.h" somewhere.

opinion?

RudolfAtHome, 2021-06-04:
Yes, I am using this extension and it works.
I was looking for this library because it works on my 3 wifi enabled controllers!
These are Nano 33 IOT, Nano RP2040 and the ESP family.

RobTillaart, 2021-06-04:
Good to hear the lib works on those different platforms. I mainly tested UNO and ESP32.

Had some thoughts about fixing this issue in the lib but it should be fixed in Arduino.h (or one of its dependencies).
So for now I am not going to add it to the library.

closed 2021-06-04 by RobTillaart