RobTillaart/DHTNew #71: AM2315

JopWerff, 2022-01-05:
Hoi Rob,

Your DHTNew is a very nice library and has advantages over the other DHT-libraries. I tested it with the DHT22.
I'm working with an ESP32,

For my real life project in the open air in conditions of rain and frost I would like to use a more robust version.
This one:
[https://www.tinytronics.nl/shop/nl/sensoren/lucht/vochtigheid/am2315-i2c-temperatuur-en-vochtigheidsmeter](url)

This complete product is using the **AM2315**, as you see.

But  I read in the specs for your DHTNew:
"Supports DHT11, DHT22, DHT33, DHT44, **AM2301**, **AM2302**, **AM2303** as these all have the same protocol."

You will understand my question now: is your fine library DHTNew also suitable for the **AM2315**?

 Kind regards,

    Jop

RobTillaart, 2022-01-05:
Good morning Job,

Never worked with the AM2315 so I have to check the datasheet.

RobTillaart, 2022-01-05:
Hoi Job,

Sorry this is an I2C device so it cannot be controlled by the DHTNew library.
The specs (error rate) of that sensor look quite good. 

The data transferred resembles my DHT12 library which is I2C. 
It uses another fixed address (easy to patch), and the low level datatransfer might need some fixes. 
(I see the AM2315 has far more registers to be read)

You mention the DHTNew has some advantages over other libs. 
Can you specify which ones you need?

Best option is to create a new library specific for this sensor.
I'm busy this week and at least part of next but if you are in a hurry we might discuss a sponsored project.

Regards,
Rob

RobTillaart, 2022-01-05:
An alternative might be - https://github.com/adafruit/Adafruit_AM2315
It is basic but it will get you started.

JopWerff, 2022-01-05:
Hoi Rob,

Thnx for your lightning fast response.

To answer your last question - the advantages of your DHTNew:
  - non-blocking read of device, if seems to be faster in reading values
  - uses just one pin (my concept is getting a bit overcrowded with other sensors: 5 x DS18B20, BME280)
  - you library looks elegant, easy to implement
  
 I see your point, AM2315 is an only-I2C-sensor - your proposed DHT12 has two options, I2C and one-wire.

The Adafruit library, you mentioned, I tested already a bit.

But you wrote about your DHT12-library. Where to find?
After some wandering around on your Github I found an almost secret or hidden wealth of libraries you developed.
[https://github.com/RobTillaart/Arduino/tree/master/libraries](url)

   (may you should "showcase" these libraries a bit more. )

**Is your DHT12-library really suitable for the AM2315?** 
**And a better alternative compared to the Adafruit_AM2315?**

Unfortunately there is no budget for sponsoring development of a good library, as all this work is done for an Ice-skating club in Winterswijk (WIJV), Netherlands. And they run out of budget. The only thing that I can do is mentioning you in the credits in the software for a system that is used to sensor on many places temperatures and humidity.
This natural ice rink is getting a lot of publicity (google just for the words: Winterswijk, schaatsen, ijs, natuurijs, "Wonder van Winterswijk"). 
As you know, as a Dutchman, Holland is getting crazy when there is frost and natural ice!
Here some info:
[https://www.trouw.nl/binnenland/weinig-kans-op-schaatspret-op-natuurijs-behalve-in-winterswijk~b8b5b455/](url)
[https://jeugdjournaal.nl/artikel/2410373-schaatsplezier-in-winterswijk-nu-al-natuurijs.html](url)
[https://www.nporadio1.nl/fragmenten/nieuws-en-co/e5982c44-4a2a-4718-8a9f-7e00289fb636/2021-12-21-in-winterswijk-kan-al-geschaatst-worden](url)
[https://www.hartvannederland.nl/nieuws/weer/eerste-meters-op-natuurijs-zijn-geschaatst](url)
[https://www.gelderlander.nl/achterhoek/schaatspret-op-tweede-kerstdag-in-winterswijk-we-durfden-het-niet-voor-te-stellen-aan-onze-vrouwen~afadade3/](url)

See also this attach, a newspaper picture with short description
 
[image]

Regards,

    Jop

RobTillaart, 2022-01-05:
**DHTNew advantages**
Clear points, 

**DHT12**
The DHT12 library can be found here - https://github.com/RobTillaart/DHT12
But I'm not sure if it is compatible, not read the very details yet.

**AM2315 library**
> Unfortunately there is no budget for sponsoring development of a good library, as all this work is done for an Ice-skating club in Winterswijk (WIJV), Netherlands. And they run out of budget. The [...]

OK, I understand, and a mention is fair enough. I will see what I can do this week.

What I propose is to start with the DHTNew library as interface and adapt it to meet the AM2315 specs.
(deleting code is faster than typing new)

Would you be able to test the library?

RobTillaart, 2022-01-05:
PS, no skater myself, it looks pretty cool (in both ways :) to have such skating track. 
How long is it?

RobTillaart, 2022-01-05:
Q: which processor board do you use?

RobTillaart, 2022-01-05:
> What I propose is to start with the DHTNew library as interface and adapt it to meet the AM2315 specs.

apparently I need to strip more (OK my party)

RobTillaart, 2022-01-05:
Almost created a first version of the library - https://github.com/RobTillaart/AM2315

Please let me know if it works.
If not please share the output

JopWerff, 2022-01-05:
Hoi Rob,

I'am very pleased if you will adapt the DHTNew Library. And I assume that this will uplevel  your library as suitable for more I2C-humidity sensors of this type (so using 2 pins as the DHT12 and some range in the AM-series).

I think I'm able to test. Testing in the practice of the skate rink in Winterswijk will be later.
In your DHT12 library I see useful routines as:
void setHumOffset(float offset) set an offset to calibrate (1st order) the sensor.
float getHumOffset() return current offset, default 0.
void setTempOffset(float offset) set an offset to calibrate (1st order) the sensor
float getTempOffset() return current offset, default 0.
Calibration: you should give me a tip how to calibrate for humidity.

I use this processor (12x):
[LilyGO TTGO T7 Mini32 V1.3 ESP32 - 4MB Flash](https://www.tinytronics.nl/shop/nl/development-boards/microcontroller-boards/met-wi-fi/lilygo-ttgo-t7-mini32-v1.3-esp32-4mb-flash)

I will mention your name in the credits for this software. Moreover I will attend the guys at TinyTronics (I'm a regular customer) on your libraries as very useful for DHT and also for this AM-type sensors. Maybe they will put a link to your software at their products.

The skating tracks in Winterswijk consist of a 400m-rink (official distance for competitions) and a rectangular ice field of 1000m2, meant for young children, beginners and ice dancers.

Thnx for your offer to do this adaptations,

    Jop

RobTillaart, 2022-01-05:
> Calibration: you should give me a tip how to calibrate for humidity.

Humidity is hard to measure exactly.  I had a SHT85 sensor from Sensirion which is a pretty reliable sensors. However I was able to fry it (and they are more expensive than this AM one). Currently I have two in a back order.

So you might look at - https://www.weerplaza.nl/actueel/relatieve-vochtigheid/  or similar sites to do a 1st order calibration.
You can click on some numbers there and see actual data e.g. https://www.weerplaza.nl/nederland/hupsel/2749993/actueel/

RobTillaart, 2022-01-05:
I have to switch to other work now, so it can take time before replies appear.