RobTillaart/DHTNew #72: DTH11 gets too short start impulse

arnowa, 2022-01-30:
I just tried this library with a DHT11. Did not work. A test with an oscilloscope showed a LOW pulse of just 3.3ms, while the DHT11 needs 18ms. 

The problem seems to be calling delayMicroseconds() with a value larger than 16383: https://www.arduino.cc/reference/en/language/functions/time/delaymicroseconds/

A very simple fix is (line 329 in dhtnew.cpp):

  delayMicroseconds(_wakeupDelay * 1100UL);
->   
  delayMicroseconds(_wakeupDelay * 550UL);
  delayMicroseconds(_wakeupDelay * 550UL);

RobTillaart, 2022-01-30:
Thank you for reporting plus providing a simple solution.

RobTillaart, 2022-01-30:
Which processor do you use?
Which IDE version/ develop platform?
Thanks

RobTillaart, 2022-01-31:
There is a related issue with the **delayMicroseconds(18000)**. 
Some applications using RTOS need to call **yield()** for context switching, e.g. to handle a Wifi incoming buffer etc. 
So the solution will be something more "complicated" than just splitting the call.

RobTillaart, 2022-01-31:
@arnowa 
Created a Pull Request https://github.com/RobTillaart/DHTNew/pull/73 with a rewrite of the wake-up delay. 
Please verify if this fix works for you.

Replaced the   ```delayMicroseconds(_wakeupDelay * 1100UL);``` with a loop that calls **yield()** every iteration.
For a DHT22 alike it will be called only once, for the DHT11 alike it will be called 18 times.

```cpp
  // WAKE UP - add 10% extra for timing inaccuracies in sensor.
  uint32_t startWakeup = micros();
  do
  {
    // HANDLE PENDING IRQ
    yield();
    delayMicroseconds(1100UL);
  }
  while((micros() - startWakeup) < (_wakeupDelay * 1100UL));
```

Non-RTOS environments (e.g. UNO) implement yield() as an empty function which will be optimized away.

arnowa, 2022-01-31:
>    Created a Pull Request [2]#73 with a rewrite of the wake-up delay.
>    [6]Android.
> References

Sure, will test it today or tomorrow and let you know.

Arno

arnowa, 2022-01-31:
>    [4]Android.

Hi Rob,

plain cheapduino Uno R3 clone (i.e. ATMega 328)
and latest Arduino IDE 1.8.19 on 64 bit Linux.

Regards,
Arno

arnowa, 2022-01-31:
Ok, works. 

Some comments:

I thinkt that

   (micros() - startWakeup) < (_wakeupDelay * 1100UL)

will promote all arguments correctly to unsigned long. 
Hence the calculation will be correct even if micros() 
wraps as (micros() - startWakeup) will wrap right 
back in that case.

Whether a yield() every 1.1ms is enought when a 19.8ms
delay is too long is something I cannot answer.

RobTillaart, 2022-01-31:
Good point, could make it 550.

RobTillaart, 2022-01-31:
@arnowa 

Which board do you use?
Which IDE?

arnowa, 2022-01-31:
plain cheapduino Uno R3 clone (i.e. ATMega 328) and latest Arduino IDE 1.8.19 on 64 bit Linux.

arnowa, 2022-01-31:
>    Thanks.
>    Did some math and every 360 us would match 18 x 1100 = 19.800 exactly.
>    Given the loop has some overhead it would at most 'overflow' 359 us.
>    3x better than 1100.

I would go with that and see whether anybody runs into trouble.
As this is also better than the original 1.1ms for the DHT22, 
it should not cause any new problems.

RobTillaart, 2022-01-31:
> plain cheapduino Uno R3 clone (i.e. ATMega 328) and latest Arduino IDE 1.8.19 on 64 bit Linux.

Ok that allows me to recreate/test a bit

RobTillaart, 2022-01-31:
Did some more math/testruns and came to 180.

For DHT11 ==> 19808 us
For DHT22 ==> 1104 us

so quite a perfect .

```cpp
  // WAKE UP - add 10% extra for timing inaccuracies in sensor.
  uint32_t startWakeup = micros();
  do
  {
    // HANDLE PENDING IRQ
    yield();
    delayMicroseconds(180UL);
  }
  while((micros() - startWakeup) < (_wakeupDelay * 1100UL));
```

RobTillaart, 2022-01-31:
> The problem seems to be calling delayMicroseconds() with a value larger than 16383: https://www.arduino.cc/reference/en/language/functions/time/delaymicroseconds/

The parameter of delayMicros() is **unsigned int** that implies one could go up to 65535
(it is hand-tuned code, so I stop digging)

closed 2022-01-31 by RobTillaart

RobTillaart, 2022-01-31:
Solution merged in master

commit c614de47 (2022-01-31): Fix #72 delayMicroseconds on wakeup
commit d0dc1c4b (2022-01-31): Fix #72 delayMicroseconds on wakeup
commit 2a42ee53 (2022-01-31): Fix #72 delayMicroseconds on wakeup (#73)

* Fix #72 delayMicroseconds on wakeup