RobTillaart/DHTNew #74: setTempOffset and setHumOffset

us292231, 2022-02-28:
It looks like the header has definitions for setTempOffset/getTempOffset and setHumOffset/getHumOffset and the class has the variables, but it doesn't seem that the implementation is missing.

us292231, 2022-02-28:
Ok, I'm an idiot.  I see it now...

closed 2022-02-28 by us292231

RobTillaart, 2022-03-01:
Q: do I need to improve documentation, comments or readme.md ?

us292231, 2022-03-01:
Hi,

Your code and documentation are really nice, no improvements needed.  Also, a big thank you for putting it out there, it has really helped me.  Your efforts are very much appreciated.

I made a really bad mistake and jumped to a conclusion that was wrong.  I forgot to connect the dots in my code and didn't catch that until just after I clicked save...  Totally my bad in multiple ways.

Thanks for asking,
Jim