RobTillaart/DHTNew #79: Si7021 by Sonoff

phiguimaran, 2022-10-12:
Hi, I'm usinng your library in various proyects, it's great !!
I'm trying to use it with the sensor that's came with Sonoff TH. 
It's a board that include Si7021 sensor and some circuits to comunicate with 1 wire.
To make this work I have to change the wakeup delay to a value near 500 microseconds
If I coment lines :
`  do
  {
    // HANDLE PENDING IRQ
    yield();
    // 180 gives good wakeup delay on UNO for DHT22 / DHT11 (issue #72)
    delayMicroseconds(180UL);  
  }
  while((micros() - startWakeup) < (_wakeupDelay * 1100UL));
`
and change for delayMicroseconds(500UL);  its work well.

I wonder if there is a way to configure this without having to edit the library?

Thanks in advance

RobTillaart, 2022-10-13:
Thanks for the issue. 
I will investigate later today.

RobTillaart, 2022-10-13:
@phiguimaran 
Pushed initial version of experimental Si7021 support. (not tested as I have no such sensor)

I added the SI7021 to the auto detect part in **read()**.

Please download the develop branch and test if this sensor detection works.
You need to add ```Serial.println(sensor.getType())```  after the **read()** function.

If detection does not work:
- force the type to 70 by using the next line in **setup()**
```cpp
sensor.setType(70);
```

phiguimaran, 2022-10-13:
Good morning, and thanks for the fast update.
I tested and the type autodetection works well.
The problem is the total delay now is 720 (because 180x3 < 550 so the next iteration add 180 micros more)
With 720 this sensor doesent work.
Let me test delay min and max that makes the sensor works well and see if it can be a 180 multiple
I'll tell you the results in a bit

RobTillaart, 2022-10-13:
Good afternoon (15:30 here) 
I can patch those numbers relative easily, only thing is that delayMicroseconds() uses a multiple of 4 on AVR.
Can reduce to 40  instead of 180.

phiguimaran, 2022-10-13:
Perfecto I need a time for test and determine a stable wakeupdelay multiple of 40
I write you later

phiguimaran, 2022-10-14:
Well, I make a lot of test and de only way I can use the Sonoff Si7020 stably is the next:
line 71: #define DHTLIB_SI7021_WAKEUP          (500UL)
line 233:   if ((_type == 22) || (_type == 70)) // DHT22, DHT33, DHT44, compatible
line 360:
if (_type==70){
	delayMicroseconds(500);  }
else {
  original loop
}

This board is extremely sensitive about wekeup time.

RobTillaart, 2022-10-14:
Thanks for your extensive testing,
I will fix in the library as you proposed. Will be done within 15 minutes. 

Need to order such sensor myself to do more testing.

RobTillaart, 2022-10-14:
Branch pushed, build is running

RobTillaart, 2022-10-15:
@phiguimaran
I've updated the code to your proposal  - used the **#define DHTLIB_SI7021_WAKEUP**

Other changes do not affect the code
- Add RP2040 pico in build-CI
- Add CHANGELOG.md  => removed history from .cpp file

Unless there are problems I will merge later today.

RobTillaart, 2022-10-15:
@phiguimaran 
Merged develop branch into master, feel free to reopen this issue if needed.

closed 2022-10-15 by RobTillaart

RobTillaart, 2022-10-16:
@phiguimaran
FYI https://github.com/RobTillaart/DHTNew/issues/81

phiguimaran, 2022-10-17:
Hi Rob, I was away for the weekend. 
You made the changes very quickly :-)
I can't test it today but y see line #183
  if (_type == 22)    //  DHT22, DHT33, DHT44, compatible
should be
 if ((_type == 22) || (_type == 70)) // DHT22, DHT33, DHT44, compatible

otherwise we get wrong readings

best regards

RobTillaart, 2022-10-17:
From - https://github.com/RobTillaart/DHTNew/issues/81

A DHT22 got recognized as a type 70 and still got good data. 

Need to look at the code to see if I understand what is happening.

RobTillaart, 2022-10-17:
Yes, understand the issue

(exactly the reason testing is so important).

RobTillaart, 2022-10-17:
TOPIC is continued in #84

commit 11e0afac (2022-10-14): fix #79 wakeup timing Si7021 (experimental)