RobTillaart/DHTNew #81: DHT22 registers as type 70 in 0.4.14

simonbogh, 2022-10-16:
DHT22 registers as type 70 in `0.4.14`. Works as intended in `0.4.13` where it registers as type 22.

RobTillaart, 2022-10-16:
Thanks for this issue, I will look into it asap.

What processor do you use?
does the DHT22 give correct readings?

simonbogh, 2022-10-16:
Thanks. 

I use an esp-wroom-32, the board is a Wemos Lolin32 ESP32 OLED.

RobTillaart, 2022-10-16:
OK, so the problem is the automatic recognition of the type.

simonbogh, 2022-10-16:
Looks like it. If I use `mySensor.setType(22);` in `0.4.14` it gives correct readings and type.

RobTillaart, 2022-10-16:
DHTNEW.cpp
```cpp
int DHTNEW::read()
{
  ...
  //  AUTODETECT

  _type = 22;
  _wakeupDelay = DHTLIB_DHT_WAKEUP;
  int rv = _read();
  if (rv == DHTLIB_OK) return rv;

  _type = 11;
  _wakeupDelay = DHTLIB_DHT11_WAKEUP;
  rv = _read();
  if (rv == DHTLIB_OK) return rv;

  //  experimental 0.4.14
  _type = 70;
  _wakeupDelay = DHTLIB_SI7021_WAKEUP;
  rv = _read();
  if (rv == DHTLIB_OK) return rv;

  _type = 0; // retry next time
  return rv;
}
```

The code tries the DHT22 before the Si7021, so the detection order is OK too.

RobTillaart, 2022-10-16:
@simonbogh 

Can you in your code before the very first call to DHT.read() add a delay of 2 seconds?

Assumption is that the sensor needs time to wake up and that it just has not had enough time for it.
Thus failing type == 22 , tries type == 11 - which fails too and finally succeeds in being a type 70.

RobTillaart, 2022-10-16:
Tried adding a delay(1000) in the auto detect an the SNR error disappeared in **endless.ino**.

So I assume this (first) error already existed in the 0.4.13 and before. Lets test.

simonbogh, 2022-10-16:
That does the trick. Even a small delay of 500 ms works on my end 👍

RobTillaart, 2022-10-16:
Created a develop branch, added a delay of max 1 second in first call

~line 123 dhtnew.cpp

added 
```cpp
  //  AUTODETECT
  //  make sure sensor had time to wake up.
  while (millis() < 1000);
```

RobTillaart, 2022-10-16:
Going to merge the fix.

closed 2022-10-16 by RobTillaart

RobTillaart, 2022-10-16:
@simonbogh 
Thanks!

commit 501cd07e (2022-10-16): fix #81 - add wakeup delay in read() (#82)

* add a wait loop in read() to give the sensor time to wake up.