RobTillaart/DHTNew #83: Support for DS18B20

simonbogh, 2022-10-17:
Not sure if it makes sense as it is not from the DHT family, but support for DS18B20 would be great so that we can use just one library. Just a thought.

RobTillaart, 2022-10-17:
The protocol is completely different. 
Also Ds18 does no humidity.

simonbogh, 2022-10-17:
True. You can put it in a backlog or delete 😊

closed 2022-10-17 by simonbogh

RobTillaart, 2022-10-17:
It is an idea long ago already discussed on the forum IIRC.

Make a base class temperature sensor  T
Derive all temperature only classes from T, so as they implement the base class, they have the same interface
Derive a base class temp+humi sensor  TH
Derive all temp+humi classes from TH, so they also have similar interfaces.

problem is that for beginners this could be too complex, and it only works well if all library builders would agree.
Furthermore you have really different protocols that might interfere.

It is however a great exercise for students in a computer class to redesign such sensor hierarchy.