RobTillaart/DHTNew #84: Sonoff Si7021 gives wrong temperature and humidity

RobTillaart, 2022-10-17:
Hi Rob, I was away for the weekend. 
You made the changes very quickly :-)
I can't test it today but y see line #183
  if (_type == 22)    //  DHT22, DHT33, DHT44, compatible
should be
 if ((_type == 22) || (_type == 70)) // DHT22, DHT33, DHT44, compatible

otherwise we get wrong readings

best regards

_Originally posted by @phiguimaran in https://github.com/RobTillaart/DHTNew/issues/79#issuecomment-1280807558_

RobTillaart, 2022-10-17:
From - https://github.com/RobTillaart/DHTNew/issues/81

A DHT22 got recognized as a type 70 and still got good data.

Need to look at the code to see if I understand what is happening.

----
update

Yes, understand the issue. I will create a new develop branch.

(exactly the reason why testing is so important).

RobTillaart, 2022-10-17:
@phiguimaran

I implement it slightly different.  I test for type == 11 first, so I do one test less.

RobTillaart, 2022-10-17:
@phiguimaran
Can you check - https://github.com/RobTillaart/DHTNew/tree/develop

phiguimaran, 2022-10-22:
Hi, sorry for the delay. I was busy with other projects.
I test your library today and it's work perfect for the Sonoff Si7021 and for the DHT21/AM2301 that I have.
Here you can see the readings with the Sonoff sensor https://thingspeak.com/channels/1505920
Thanks for the changes you made, now I can use your library without modifications.

commit 5142fe94 (2022-10-22): fix #84 Si7021 (#85)

* fix #84 improve support for Si7021
* fix keywords.txt
* minor edits
closed 2022-10-22 by RobTillaart