RobTillaart/DHTNew #86: Compiler warning, conversion float to double

pabloandresm, 2022-11-05:
Hello there,

Compiling with extra warning flags I've noticed an annoying warning (that even brakes compilation if -Werror).

.....\libraries\DHTNEW\dhtnew.cpp: In member function 'int DHTNEW::_read()':
.....\libraries\DHTNEW\dhtnew.cpp:243:18: warning: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Wdouble-promotion]
   if (_humOffset != 0.0)
       ~~~~~~~~~~~^~~~~~
.....\libraries\DHTNEW\dhtnew.cpp:249:19: warning: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Wdouble-promotion]
   if (_tempOffset != 0.0)
       ~~~~~~~~~~~~^~~~~~

Comparing against '(float)0.0' instead of '0.0' should clear that warning.

Thanks

RobTillaart, 2022-11-05:
Good afternoon Pablo,

Thanks for the issue, I will try to fix it this weekend (busy on another bug now)

Which platform do you use?

pabloandresm, 2022-11-05:
I use ESP32 with the Arduino IDE.

I always try to use -Wall -Werror and even --pedantic :), as the warnings that are reported are sometimes important. I advise my friends to compile with all of that too :)

Thank you Rob

RobTillaart, 2022-11-05:
@pabloandresm 
Created a develop branch, no sensor here nearby to do a quick test.

pabloandresm, 2022-11-05:
Well there is not much test to do. The 'if (_tempOffset != (float)0.0)' will not impact anything, it will just avoid the warning message during compilation. Without the '(float)' the 0.0 will be taken as double and hence issue the warning.

Strangely as it may seem, the warning does not appear when, for example, assigning _tempOffset  a 0.0, but just at comparison in those 2 lines I mentioned before, 243 and 249.

RobTillaart, 2022-11-05:
mmm, that is how compilers work, they look different at comparisons and assignments. With an assignment of a constant to a variable of a smaller type the compiler can deduce it will fit. When comparing with a variable it cannot know if there is a range problem. 
I did cast all assignments too, to remove all ambiguity with 0.0 from the code. 

I will merge in a few minutes.

closed 2022-11-05 by RobTillaart

RobTillaart, 2022-11-05:
Merged the fix, so if there are other issues, let me know.

pabloandresm, 2022-11-05:
That was quick! Thank you very much.

commit fec40b76 (2022-11-05): fix #86: (float)0.0 (#87)