RobTillaart/DHTNew #88: Ticker compatibility

vsmaldino, 2022-12-25:
Wemos D1R2 - ver. 0.4.17
The Library doesn't work if used within a function called by a Ticker, the board generates an exception and restarts.
Output:
`dhtnew_test.ino
LIBRARY VERSION: 0.4.17

1. Type detection test, first run might take longer to determine type
STAT	HUMI	TEMP	TIME	TYPE
OK,	77.5,	19.7,	5270,	22

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:137 __yield

>>>stack>>>
(((((cut)))))`

Code:
`#include <Ticker.h>
#include <dhtnew.h>

Ticker tckRead;
DHTNEW mySensor(D5); // ESP 16    UNO 5    MKR1010 5

void setup()
{
  Serial.begin(115200);
  delay(5000);
  Serial.println("dhtnew_test.ino");
  Serial.print("LIBRARY VERSION: ");
  Serial.println(DHTNEW_LIB_VERSION);
  Serial.println();

  Serial.println("\n1. Type detection test, first run might take longer to determine type");
  Serial.println("STAT\tHUMI\tTEMP\tTIME\tTYPE");
  test();
  delay(1000);
  tckRead.attach(3.0, test);
} // setup

//
void loop()
{
  delay(1);
} // loop

//
void test()
{
  // READ DATA
  uint32_t start = micros();
  int chk = mySensor.read();
  uint32_t stop = micros();

  switch (chk)
  {
  case DHTLIB_OK:
    Serial.print("OK,\t");
    break;
  case DHTLIB_ERROR_CHECKSUM:
    Serial.print("Checksum error,\t");
    break;
  case DHTLIB_ERROR_TIMEOUT_A:
    Serial.print("Time out A error,\t");
    break;
  case DHTLIB_ERROR_TIMEOUT_B:
    Serial.print("Time out B error,\t");
    break;
  case DHTLIB_ERROR_TIMEOUT_C:
    Serial.print("Time out C error,\t");
    break;
  case DHTLIB_ERROR_TIMEOUT_D:
    Serial.print("Time out D error,\t");
    break;
  case DHTLIB_ERROR_SENSOR_NOT_READY:
    Serial.print("Sensor not ready,\t");
    break;
  case DHTLIB_ERROR_BIT_SHIFT:
    Serial.print("Bit shift error,\t");
    break;
  case DHTLIB_WAITING_FOR_READ:
    Serial.print("Waiting for read,\t");
    break;
  default:
    Serial.print("Unknown: ");
    Serial.print(chk);
    Serial.print(",\t");
    break;
  }
  // DISPLAY DATA
  Serial.print(mySensor.getHumidity(), 1);
  Serial.print(",\t");
  Serial.print(mySensor.getTemperature(), 1);
  Serial.print(",\t");
  uint32_t duration = stop - start;
  Serial.print(duration);
  Serial.print(",\t");
  Serial.println(mySensor.getType());
  // delay(2500);
} // test

// -- END OF FILE --
`

RobTillaart, 2022-12-25:
Thanks for the issue,
I will look at it asap but that might be next year 

I have no experience with ticker, do you have a link that explainis what it is / how it works / repo ?

RobTillaart, 2022-12-25:
Can you compile and run the examples?

RobTillaart, 2022-12-25:
https://github.com/sstaub/Ticker
This one?

vsmaldino, 2022-12-25:
> Thanks for the issue, I will look at it asap but that might be next year
> 
> I have no experience with ticker, do you have a link that explainis what it is / how it works / repo ?

For Ticker you can start here:
https://techtutorialsx.com/2021/08/07/esp32-ticker-library/
https://circuits4you.com/2018/01/02/esp8266-timer-ticker-example/

vsmaldino, 2022-12-25:
> Can you compile and run the examples?

Yes sure. If you look at my code, "setup" calls the "test" and it works fine (see the output). The problem is when "test" is called by the ticker.

vsmaldino, 2022-12-25:
> https://github.com/sstaub/Ticker This one?

I'm using the standard Ticker library

RobTillaart, 2022-12-26:
> > https://github.com/sstaub/Ticker This one?
> 
> I'm using the standard Ticker library

Where can I find the standard Ticker?

vsmaldino, 2022-12-26:
> > > https://github.com/sstaub/Ticker This one?
> > 
> > 
> > I'm using the standard Ticker library
> 
> Where can I find the standard Ticker?

It is included in the "ESP8266 Boards" kit I imported within Arduino IDE using "Boards Manager"

RobTillaart, 2022-12-28:
Did some reading in the library code (no experience with esp8266) 

Can you try **  tckRead.attach_scheduled(3.0, test);** instead of **  tckRead.attach(3.0, test);**

The comments states different context. 
The CTX context fails so the idea is to try another (to better understand the detailed working).
```cpp
    // callback will be called at following loop() after ticker fires
    void attach_scheduled(float seconds, callback_function_t callback)
    {
        _callback_function = [callback]() { schedule_function(callback); };
        _attach_ms(1000UL * seconds, true);
    }

    // callback will be called in SYS ctx when ticker fires
    void attach(float seconds, callback_function_t callback)
    {
        _callback_function = std::move(callback);
        _attach_ms(1000UL * seconds, true);
    }
```

RobTillaart, 2023-01-09:
@vsmaldino 
did you test the above ?

RobTillaart, 2023-01-12:
As I got no response I close the issue

closed 2023-01-12 by RobTillaart