RobTillaart/DHTNew #91: consider an enhancement to be able to compile this lib directly under espressif/platformio without Arduino layer

pabloandresm, 2023-03-13:
Title says it all. It would be nice.

As the library only uses pinMode() and digitalWrite(), a wrapper for those for ESP32 are pretty much straightforward:

They can even be taken from the Arduino/ESP32 libs at hardware/esp32/2.0.7/cores/esp32/esp32-hal-gpio.c

Then a modification in the library.json for the "frameworks":"*" et voila! :)

RobTillaart, 2023-03-14:
Thanks for the issue,

I do not use platformIO and ESP32 only seldom. I'm on the 1.0.6 version for ESP32 under IDE1.8.19

do you think of a wrapper something like 
```cpp
inline void PM(pin, mode)
{
#ifdef ESP32
__pinMode(pin, mode)
#else
pinMode(pin, mode)
}

inline void DW(pin, value)
{
#ifdef ESP32
__digitalWrite(pin, mode)
#else
digitalWrite(pin, mode)
}

inline uint8_t DR(pin)
{
#ifdef ESP32
return __digitalRead(pin)
#else
return digitalRead(pin)
}
```

Is it possible for you to create a PR?

pabloandresm, 2023-03-14:
Hi Rob for your answer.

I was thinking more of something like:

ifdef esp-idf, and not arduino
pinMode() { }
digitalWrite() { }
endif

So the code is not touched at all.

I already compiled your lib for espressif only. Let me clean my code up and prepare a proper PR.