RobTillaart/DHTNew #97: AM2322 Humidity 99.9 Temp OK

pacmac, 2024-04-09:
Hi;

Am testing this lib on the AM2322, I have a bunch of them and are testing on 2 of them, both give same results.

I have tried different pins (ESP32) but the results are the same on both devices and different pins.

Temperature reading is good, but humidity is always 99.9, if I introduce a negative offset then the values will always be 99.9 - that offset.

Have tried defining type as 22 or not defining type, result is the same, have also tried slowing down the reads to 5 seconds but no difference.

Any suggestions ?

RobTillaart, 2024-04-10:
Thanks for the issue.
Might take a few days to dive into it.

RobTillaart, 2024-04-10:
If I recall correctly the device also supports I2C?
Have you tried that interface to see if humidity sensor does work? 
See AM232X library.

RobTillaart, 2024-04-10:
Can you run the sketch **dhtnew_pulse_diag.ino**
and post its output?

pacmac, 2024-04-10:
Thanks, could not figure out why I was getting no output, then realised had to Serial.begin before waiting for it:

```
void setup()
{

  Serial.begin(115200);
  while(!Serial);        //  MKR1010 needs this

  Serial.println("dhtnew_pulse_diag.ino");
  Serial.println();

  //  default should be HIGH
  pinMode(_dataPin, OUTPUT);
  digitalWrite(_dataPin, HIGH);
}

```

Here's the output:

```
15:17:57.499 -> awake 2 3 4 5 6 7 
15:17:57.540 -> 
15:17:57.540 -> RUN:	9
15:17:57.540 -> IDX:	89
15:17:57.540 -> WAKEUP
15:17:57.540 -> 	1130	14	29	81	64	18
15:17:57.540 -> HUM
15:17:57.540 -> 	37	25	54	26	54	25	55	25	55	25	55	25	55	73	54	72
15:17:57.540 -> 	68	73	55	72	55	73	55	25	54	26	54	73	55	73	54	78
15:17:57.540 -> 
15:17:57.540 -> TEMP
15:17:57.540 -> 	62	26	54	26	54	26	54	26	54	26	54	26	54	26	54	73
15:17:57.540 -> 	66	27	53	26	54	26	55	73	54	26	54	26	54	26	54	73
15:17:57.540 -> 
15:17:57.540 -> CRC
15:17:57.540 -> 	64	74	53	74	54	74	54	73
15:17:57.540 -> 	55	73	54	74	54	26	54	42
15:17:57.540 -> 
15:17:57.540 -> BYE
15:17:57.540 -> 	46	1
15:17:57.540 -> 
```

RobTillaart, 2024-04-10:
Thanks

[image]

RobTillaart, 2024-04-10:
Also this section of the readme might be the cause.

Can you check your power supply?

___Voltage AM2301 and ESP8266__
In a test an AM2301 had problems giving no humidity (99.9% overflow) when the DHTStable library was used with an ESP8266. (Reported by mail, no GH issue). As this DHTStable library is strongly related to the DHTNew it is mentioned here too._

_After days of testing and thinking and more testing the cause was found. The AM2301 was powered by a 5V3 power supply which was apparently too high while having the data handshakes at 3V3. When the VCC voltage was lowered to 5V1 it appeared to work as it should. (Kudos to Viktor for finding the cause)_

RobTillaart, 2024-04-10:
English datasheet - https://robu.in/wp-content/uploads/2022/06/1290199_AM2322-Datasheet.pdf

[image]

pacmac, 2024-04-10:
I am running off 3V3 device is Adafruit QT PY S3.

I have added back the pullup as I had to switch device due to other one being on battery drain test.

```
18:45:38.999 -> awake 2 3 4 5 6 7 
18:45:39.032 -> 
18:45:39.032 -> RUN:	2
18:45:39.032 -> IDX:	89
18:45:39.032 -> WAKEUP
18:45:39.032 -> 	1132	14	28	81	64	18
18:45:39.032 -> HUM
18:45:39.032 -> 	35	27	53	26	54	26	54	26	54	26	54	26	54	73	54	73
18:45:39.032 -> 	67	74	54	73	54	74	53	26	54	26	54	74	54	73	54	25
18:45:39.032 -> 
18:45:39.032 -> TEMP
18:45:39.032 -> 	68	25	55	25	55	25	54	26	54	26	54	26	54	26	54	73
18:45:39.032 -> 	66	27	53	27	53	74	54	26	54	73	54	26	54	74	53	74
18:45:39.032 -> 
18:45:39.032 -> CRC
18:45:39.032 -> 	64	26	54	26	54	25	55	73
18:45:39.032 -> 	54	26	53	75	53	26	54	90
18:45:39.032 -> 
18:45:39.032 -> BYE
18:45:39.032 -> 	46	1
18:45:39.032 -> 
```

Strange that the temperature figures are good !

RobTillaart, 2024-04-10:
> I am running off 3V3 device is Adafruit QT PY S3.

OK, so that cannot be the cause

> Strange that the temperature figures are good !

No, that is what is expected, strange the humidity fails. 
However the first bit (35  27) is very short, that is the only clue so far. 
Need to dive into the code to see if that triggers something..

What value does the **dht.read()** function return, it might give another clue .

pacmac, 2024-04-10:
result of read():

```
19:02:25.586 -> Started
19:02:25.586 -> BEFORE OFFSET
19:02:30.552 -> 	29.3
19:02:30.750 -> 99.9
19:02:30.750 -> AFTER -10 HUMID OFFSET
19:02:30.750 -> read(): -8
19:02:32.857 -> 89.9	29.3
19:02:34.959 -> 89.9	29.3
```

RobTillaart, 2024-04-10:
Can you add

Dht.setWaitForReading(true);

After dht.begin()

RobTillaart, 2024-04-10:
Read does check if there are at least 2 seconds past since startup or last call to read.

pacmac, 2024-04-10:
I dont have mySensor.begin();

```
void setup(){

  mySensor.setType(22);
  Serial.println("BEFORE OFFSET");
  delay(5000);
  mySensor.read();
  
  Serial.print("\t");
  Serial.println(mySensor.getTemperature(), 1);
  delay(200);
  Serial.println(mySensor.getHumidity(), 1);
  mySensor.setHumOffset(-10);
  // mySensor.setTempOffset(-3.5);

  Serial.println("AFTER -10 HUMID OFFSET");

  Serial.println("read(): " + String(mySensor.read()));
}
```

If I add it before setType I get:

```
mySensor.begin();
            ^~~~~

exit status 1

Compilation error: 'class DHTNEW' has no member named 'begin'
```

RobTillaart, 2024-04-10:
Could not find a cause in the code, so I reread this whole issue again for clues.

And yes, made a mistake with my hex math ==> should not do that from the head

```
HUM
37	25	54	26	54	25	55	25	55	25	55	25	55	73	54	72
68	73	55	72	55	73	55	25	54	26	54	73	55	73	54	78

decoding every 2nd time =>   00000011 11100111  == 0x03E7 == 999 ==> 99.9%
```

So the sensor itself did sent the 99.9, so that means scrutinizing the datasheet.

RobTillaart, 2024-04-10:
> It works with different lib on I2C but I am desperately short on pins :-)

Considered an I2C multiplexer? TCA9548?

pacmac, 2024-04-11:
Hi Rob;

Thanks for all the help, the device works fine on I2C using for example the adafruit lib, both temperature and humidity are correct, I dont think the devices are defective.

RobTillaart, 2024-04-11:
Ok,
The I2C working means hat (1) T and H works.and (2) I2C is implemented correctly... That is good.news and bad news.

Explaining the bad news:
My worries are that the one wire implementation is not within specs and (after revisiting my hex math) the device is sending an invalid value. Especially because I2C is working shows that the physics part for H is OK but the one wire part of the device is suspect. The library does extract the sent percentage correctly even if the protocol is outside its specs.

.

pacmac, 2024-04-11:
Rob, thanks for your help, I guess this one just wont work.

I will have to revert to using I2C, but thanks so much again

PeterC

RobTillaart, 2024-04-11:
You're welcome.
For now I close this issue, if new insights pop up feel free to reopen it or just add comments.

closed 2024-04-11 by RobTillaart

pacmac, 2024-04-11:
Thanks again for trying.