RobTillaart/DHTNew #100: Negative temp is calculated wrongly

Fihl, 2024-11-22:
It's getting cold out there.. 

When receiving _bits[2..3] = 0xffff as 0.1 degree, it gets converted to -3200 something, because 
0x7fff is taken as 32000, and MSB of 0x8000 is taken as signbit and now makes -32000. 
Finally _temperature = -32000 * 0.1 = -3200

Code **has** to take 16 bits "as is", and move it directly into a int16_t. Then every is "as send" from DHT. And correctly interpreted. 
And no problem with -0.0, as minus-0 does not exists i the integer world. 
Code below also do the same with _humidity, but was correct before as long as DHT does not send negative values. 

My fixed code in dhtnew.cpp: 
  else               //  DHT22, DHT33, DHT44, compatible + Si7021
  {
    int16_t t1 = (_bits[0] << 8) + _bits[1]; // 16 bits, as raw int
    _humidity = t1 * 0.1;
    int16_t t2 = (_bits[2] << 8) + _bits[3]; // 16 bits, as raw int
    _temperature = t2 * 0.1;
  }

DHT11 code above also looks strange to my eyes. But I have not looked up the specs, and I do not use that part anyway. 

/Christen

RobTillaart, 2024-11-22:
Thanks for reporting the issue.

This sounds serious given the library has been around for years.
Need to investigate the datasheet and find a free sensor.

Q: which sensor do you use? DHT22 or compatible ?
Q: have you tested the sensor with another DHT22 library e.g. Adafruit DHT22 library to rule out a broken sensor?

Might take a few days as I'm quite busy, but labeled it as bug to give priority.

----

> DHT11 code above also looks strange to my eyes. But I have not looked up the specs, and I do not use that part anyway.

DHT11 does not support negative numbers, and the coding differs indeed.

RobTillaart, 2024-11-22:
> It's getting cold out there..

Which country? 
Here in Netherlands its just around freezing.

Fihl, 2024-11-22:
Version 0.4.21 (latest on Arduino 1.8), using ESP32 C3 Super Mini. DHT says it is "type 22", and also bought as DHT22 on ali. 

Denmark here and now +1, but still 10cm snow. On sunday it is again 13 degrees! 
So I made a run in the freezer, and shows below all 6 bytes in _bits[] 

I only blame the code, as the dht in fact does return 0xff-something, like 0xffff for minus 1 that gives -0.1 after times 0.1. 

Manual test would be after real reading to reset _bits[] to known values, like
`  _bits[2] = 0xff;    _bits[3] = 0x9c; // => -10.0 degrees`

My code (after making _bits[] public)
```
  float HumiVal = dht.getHumidity();
  float TempVal = dht.getTemperature();
  Serial.printf(F("(dht22: %02x %02x %02x %02x %02x temp=%.1f, humi=%.1f)"),
             dht._bits[0],dht._bits[1],dht._bits[2],dht._bits[3],dht._bits[4], TempVal, HumiVal);
```

Output with my code version. 
DHT model: 22
```
(dht22: 02 2a 00 57 83 temp=8.7, humi=55.4)
(dht22: 02 0e 00 50 60 temp=8.0, humi=52.6)
(dht22: 01 d8 00 48 21 temp=7.2, humi=47.2)
(dht22: 01 9d 00 40 de temp=6.4, humi=41.3)
(dht22: 01 62 00 37 9a temp=5.5, humi=35.4)
(dht22: 01 3d 00 2f 6d temp=4.7, humi=31.7)
(dht22: 01 2a 00 27 52 temp=3.9, humi=29.8)
(dht22: 01 1b 00 1e 3a temp=3.0, humi=28.3)
(dht22: 01 10 00 16 27 temp=2.2, humi=27.2)
(dht22: 01 12 00 0d 20 temp=1.3, humi=27.4)
(dht22: 01 11 00 06 18 temp=0.6, humi=27.3)
(dht22: 01 14 ff ff 13 temp=-0.1, humi=27.6)       0xFF FF starts here, going down
(dht22: 01 17 ff f7 0e temp=-0.9, humi=27.9)
(dht22: 01 1b ff ef 0a temp=-1.7, humi=28.3)
(dht22: 01 22 ff e8 0a temp=-2.4, humi=29.0)
(dht22: 01 28 ff e2 0a temp=-3.0, humi=29.6)
(dht22: 01 2f ff db 0a temp=-3.7, humi=30.3)
(dht22: 01 37 ff d6 0d temp=-4.2, humi=31.1)
(dht22: 01 3d ff d0 0d temp=-4.8, humi=31.7)
(dht22: 01 42 ff ca 0c temp=-5.4, humi=32.2)
(dht22: 01 48 ff c5 0d temp=-5.9, humi=32.8)
(dht22: 01 4d ff c0 0d temp=-6.4, humi=33.3)
(dht22: 01 56 ff bb 11 temp=-6.9, humi=34.2)
(dht22: 01 5c ff b6 12 temp=-7.4, humi=34.8)
(dht22: 01 65 ff b1 16 temp=-7.9, humi=35.7)
(dht22: 01 71 ff ad 1e temp=-8.3, humi=36.9)
(dht22: 01 76 ff a8 1e temp=-8.8, humi=37.4)
(dht22: 01 7b ff a3 1e temp=-9.3, humi=37.9)
(dht22: 01 81 ff a0 21 temp=-9.6, humi=38.5)
(dht22: 01 8a ff 9c 26 temp=-10.0, humi=39.4)
(dht22: 01 91 ff 99 2a temp=-10.3, humi=40.1)
(dht22: 01 98 ff 96 2e temp=-10.6, humi=40.8)
(dht22: 01 9e ff 92 30 temp=-11.0, humi=41.4)
(dht22: 01 a5 ff 8f 34 temp=-11.3, humi=42.1)
(dht22: 01 aa ff 8c 36 temp=-11.6, humi=42.6)
(dht22: 01 b0 ff 89 39 temp=-11.9, humi=43.2)
(dht22: 01 b5 ff 87 3c temp=-12.1, humi=43.7)
(dht22: 01 bb ff 84 3f temp=-12.4, humi=44.3)
(dht22: 01 c3 ff 80 43 temp=-12.8, humi=45.1)
(dht22: 01 c9 ff 7e 47 temp=-13.0, humi=45.7)
(dht22: 01 cd ff 7c 49 temp=-13.2, humi=46.1)
(dht22: 01 d3 ff 7a 4d temp=-13.4, humi=46.7)
(dht22: 01 da ff 78 52 temp=-13.6, humi=47.4)
(dht22: 01 e1 ff 76 57 temp=-13.8, humi=48.1)
(dht22: 01 e4 ff 74 58 temp=-14.0, humi=48.4)
(dht22: 01 e8 ff 73 5b temp=-14.1, humi=48.8)
(dht22: 01 ef ff 71 60 temp=-14.3, humi=49.5)
(dht22: 01 f0 ff 6f 5f temp=-14.5, humi=49.6)
(dht22: 01 f5 ff 6e 63 temp=-14.6, humi=50.1)
(dht22: 01 f8 ff 6c 64 temp=-14.8, humi=50.4)
(dht22: 01 fe ff 6b 69 temp=-14.9, humi=51.0)
(dht22: 02 00 ff 69 6a temp=-15.1, humi=51.2)
(dht22: 02 04 ff 68 6d temp=-15.2, humi=51.6)
(dht22: 02 0a ff 67 72 temp=-15.3, humi=52.2)
(dht22: 02 17 ff 63 7b temp=-15.7, humi=53.5)
(dht22: 02 1a ff 62 7d temp=-15.8, humi=53.8)
(dht22: 02 1c ff 61 7e temp=-15.9, humi=54.0)
(dht22: 02 21 ff 61 83 temp=-15.9, humi=54.5)
(dht22: 02 25 ff 5f 85 temp=-16.1, humi=54.9)
(dht22: 02 29 ff 5e 88 temp=-16.2, humi=55.3)
(dht22: 02 2c ff 5d 8a temp=-16.3, humi=55.6)
(dht22: 02 30 ff 5d 8e temp=-16.3, humi=56.0)
(dht22: 02 45 ff 56 9c temp=-17.0, humi=58.1)
(dht22: 02 5f ff 4c ac temp=-18.0, humi=60.7)
(dht22: 02 67 ff 48 b0 temp=-18.4, humi=61.5)

```
[image]

Fihl, 2024-11-22:
I did a google and others also came to conclusion that **base-2** normal layout is used, so 
t = ( highbyte <<8 ) + lowbyte is the way to read the value.  Followed by times 0.1 

I did find your dhtnew as my adafruit also failed. And one module fits all looked nice. 

Have a go at [stack xchg](https://arduino.stackexchange.com/questions/86448/dht22-sensor-reading-code-interprets-negative-values-weirdly)

Following explanation is very loose defined. (From AM2301 that issue #53 mentions)
And wrong in my mind.
[image]

RobTillaart, 2024-11-22:
[image]

Slight layout variation of the print layout

```
100	00 88 00 2C B4 		4.40	13.60
101	00 89 00 2A B3 		4.20	13.70
102	00 89 00 28 B1 		4.00	13.70
103	00 8A 00 26 B0 		3.80	13.80
104	00 8B 00 24 AF 		3.60	13.90
105	00 8C 00 22 AE 		3.40	14.00
106	00 8D 00 20 AD 		3.20	14.10
107	00 8D 00 1E AB 		3.00	14.10
108	00 8E 00 1B A9 		2.70	14.20
109	00 8F 00 19 A8 		2.50	14.30
110	00 90 00 17 A7 		2.30	14.40
111	00 91 00 15 A6 		2.10	14.50
112	00 92 00 13 A5 		1.90	14.60
113	00 92 00 11 A3 		1.70	14.60
114	00 94 00 0F A3 		1.50	14.80
115	00 94 00 0D A1 		1.30	14.80
116	00 95 00 0B A0 		1.10	14.90
117	00 96 00 0A A0 		1.00	15.00
118	00 97 00 08 9F 		0.80	15.10
119	00 98 00 06 9E 		0.60	15.20
120	00 99 00 04 9D 		0.40	15.30
121	00 9A 00 02 9C 		0.20	15.40
122	00 9C 00 00 9C 		0.00	15.60
123	00 9D 80 00 1D 		0.00	15.70
124	00 9E 80 02 20 		-0.20	15.80
125	00 9F 80 04 23 		-0.40	15.90
126	00 A0 80 06 26 		-0.60	16.00
127	00 A1 80 08 29 		-0.80	16.10
128	00 A2 80 09 2B 		-0.90	16.20
129	00 A3 80 0B 2E 		-1.10	16.30
130	00 A4 80 0C 30 		-1.20	16.40
131	00 A6 80 0E 34 		-1.40	16.60
132	00 A7 80 10 37 		-1.60	16.70
133	00 A8 80 11 39 		-1.70	16.80
134	00 A9 80 12 3B 		-1.80	16.90
135	00 AB 80 14 3F 		-2.00	17.10
136	00 AB 80 16 41 		-2.20	17.10
137	00 B2 80 17 49 		-2.30	17.80
138	00 F2 80 18 8A 		-2.40	24.20
139	01 0A 80 19 A4 		-2.50	26.60
140	01 22 80 19 BC 		-2.50	29.00
141	01 3D 80 19 D7 		-2.50	31.70
142	01 56 80 19 F0 		-2.50	34.20
143	01 70 80 19 0A 		-2.50	36.80
144	01 89 80 19 23 		-2.50	39.30
145	01 A1 80 18 3A 		-2.40	41.70
146	01 B9 80 17 51 		-2.30	44.10
147	01 D2 80 16 69 		-2.20	46.60
148	01 E9 80 14 7E 		-2.00	48.90
149	02 00 80 13 95 		-1.90	51.20
150	02 16 80 11 A9 		-1.70	53.40
151	02 2C 80 0F BD 		-1.50	55.60
152	02 41 80 0D D0 		-1.30	57.70
153	02 55 80 0B E2 		-1.10	59.70
154	02 69 80 09 F4 		-0.90	61.70
155	02 7C 80 07 05 		-0.70	63.60
156	02 8E 80 04 14 		-0.40	65.40
157	02 A0 80 00 22 		0.00	67.20
158	02 B2 00 03 B7 		0.30	69.00
159	02 C4 00 07 CD 		0.70	70.80
160	02 D5 00 0B E2 		1.10	72.50
161	02 E6 00 0E F6 		1.40	74.20
162	02 F5 00 11 08 		1.70	75.70
163	03 04 00 15 1C 		2.10	77.20
164	03 12 00 19 2E 		2.50	78.60
165	03 20 00 1D 40 		2.90	80.00
166	03 2E 00 21 52 		3.30	81.40
167	03 3B 00 25 63 		3.70	82.70
168	03 47 00 29 73 		4.10	83.90
169	03 53 00 2D 83 		4.50	85.10
170	03 5E 00 31 92 		4.90	86.20
```

So they spit out different bits when negative.

RobTillaart, 2024-11-22:
Datasheet DHT22  (does not tell how to interpret negative numbers, in fact it does not match the hardware at all)

[image]

Datasheet DHT33 (DHT44 + RHT04 shows same paragraph)

[image]

Datasheet AM2301

[image]

Datasheet CM2122

[image]

Fihl, 2024-11-22:
Wow, that is one way (your way :-) ) of sending data. 
Send value, and then add one bit to say if it is plus or minus. 
80 02 => -0.2

The other way (my way and device) sends base-2, i.e. 100% as a cpu register saves plus/minus. 
FF FE => -0.2

Seems like **you need to have a switch** saying which version each device is using. Base-2 or "strange" mode. 

The checksum says nothing about which way data is send. Just says that all 4 bytes are send correct.

RobTillaart, 2024-11-22:
@Fihl 

So the code in the library is correct but "not correct enough", two sensors both named DHT22 produce different bits.

**Q: how to solve this?**

Requirements:
1. The code should support the current negative algorithm
2. The code should support the reported negative algorithm
3. If possible the library should be able to detect which algorithm to use.

Given that the range of (negative) temperatures is something like -40 .. 80 

This means that if we look at byte 2 which is in one case 1000 0000 and in the other case 1111 1111
So the highest bit indicates negative temperatures for sure.

- minus 40 would be coded as  0x8190 
- minus 40 would be coded as  0xFE6F

So the second highest bit could be used to determine the algorithm used.

RobTillaart, 2024-11-22:
Will try to have a develop branch this evening. At the moment I am in a deadline for Monday....
Patience, breath in breath out . :)

Fihl, 2024-11-22:
I only have "my" DHT22 versions :-)

RobTillaart, 2024-11-22:
@Fihl 

Created a develop branch with a (non optimized) fix for this issue.
Please test your 50% 😎, 

Rob

Fihl, 2024-11-22:
Nice and clean
I will give it a go tomorrow.

RobTillaart, 2024-11-23:
Relates to these issues
- https://github.com/RobTillaart/DHTNew/issues/57
- https://github.com/RobTillaart/DHTNew/issues/52

RobTillaart, 2024-11-23:
@Fihl 
- found related issues in the past (see above)
- pushed a version with smaller footprint
- minor update in readme.md

Fihl, 2024-11-23:
Denmark says ok for base-2 part of code as in dev branch

Been a pleasure brainstorming with you
Christen

```
(dht22: 03 d3 00 64 3a temp=10.0, humi=97.9)
(dht22: 01 e6 00 06 ed temp=0.6, humi=48.6)
(dht22: 01 cc 00 00 cd temp=0.0, humi=46.0)
(dht22: 01 b9 ff fa b3 temp=-0.6, humi=44.1)
(dht22: 01 93 ff 4d e0 temp=-17.9, humi=40.3)
(dht22: 01 dd ff 3f 1c temp=-19.3, humi=47.7)

```

RobTillaart, 2024-11-23:
Then I will merge the dev branch and release 0.5.0

Thanks again for reporting and testing!

closed 2024-11-23 by RobTillaart

Fihl, 2024-11-23:
I guess that AM2302 might always be as documented above. 
And DHT22 might always be base-2 coded. 
My DHT22 has no claims of being a AM2302 as your picture does.

commit 08745f9d (2024-11-22): fix #100, negative representation
commit d3cff615 (2024-11-23): fix #100, negative representation (#101)

- fix #100, different negative algorithm support
- bump version