RobTillaart/DHTNew #104: 0.5.1 weird side-effect bug

mitra42, 2024-11-26:
I misspoke on 0.5.1 working :-) 

Looks like there is some kind of weird sideeffect going on. 

The code requires either `setType(11)`  *OR*  `getType()` to work.

For example - if I comment out line 83 in dhtnew_simple.ino. then it continuously reports `Sensor not ready`

I don't understand why since all getType does is return _type (which is non-zero, I checked), 

I see the same problem in my own code, if I do a setType(11) it works.

RobTillaart, 2024-11-26:
So we need more testing, and I have no hardware to do it.

The documentation mentions the ```setType()``` workaround and claims experimental so it is not a showstopper (or is it?).
labelled as bug as it is one of my most used libraries to give this issue prio.

To be continued.

mitra42, 2024-11-26:
Understood - yes, I've got a workaround so I'm good. 

It might hit other users of the library though. 

I think you'd find the same problem on your own (DHT11) hardware if you comment out that getType line in dhtnew_simple.in, it seems a program logic issue (which I can't figure out) , but I could be wrong.

RobTillaart, 2024-11-26:
My DHT11 is recognized correctly, so need to redo the recognition for the KY015

```
LIBRARY VERSION: 0.5.1

1. Type detection test, first run might take longer to determine type
STAT	HUMI	TEMP	TYPE
OK,	37.0,	20.0,	11
OK,	37.0,	20.0,	11
OK,	37.0,	20.0,	11
OK,	37.0,	20.0,	11
OK,	37.0,	20.0,	11
OK,	37.0,	20.0,	11
```

Saw some typos in the new examples too.

RobTillaart, 2024-11-26:
@mitra42 
Created a develop branch to test a slightly modified recognition code.
Please test

mitra42, 2024-11-26:
Your new code .... 
```
06:09:41.917 -> 1. Type detection test, first run might take longer to determine type
06:09:41.917 -> STAT	HUMI	TEMP	TYPE
06:09:42.871 -> Sensor not ready,	-999.0,	-999.0,	11
06:09:44.877 -> OK,	92.0,	20.7,	11
```
which looks like a faulty humidity reading, but it was 100% first time I tried, so there may be condensation on the sensor.as its first thing in the morning. 

Now - if I comment out the getType line it works
```
06:11:07.834 -> LIBRARY VERSION: 0.5.2
06:11:07.834 -> 
06:11:07.834 -> 
06:11:07.834 -> 1. Type detection test, first run might take longer to determine type
06:11:07.834 -> STAT	HUMI	TEMP	TYPE
06:11:08.755 -> Sensor not ready,	-999.0,	-999.0,	OK,	92.0,	20.7,	OK,	92.2,	20.7,
```

mitra42, 2024-11-26:
Yes - I checked by putting a comment in the getType() code itself,   that extra read() is run exactly once, after the first (typically failed) run through the loop, and after that it works correctly.

RobTillaart, 2024-11-26:
The extra _read() in line 154-158 is called to have the temperature and humidity converted correctly.
As the previous **int rv = _read();** in line 141 did not convert correctly  
That is tested in line 150  **if (_bits[0] <= 0x03)** so the type is forced to 11.

```
    //  KY015 as DHT11
    _type = 11;
    rv = _read();  //  read again with correct conversion.
    return rv;
  }
```

mmm, maybe the wakeUp delay should be set too

change
```cpp
    //  KY015 as DHT11
    _type = 11;
    rv = _read();  //  read again with correct conversion.
    return rv;
```
to
```cpp
    //  KY015 as DHT11
    _type = 11;
    _wakeupDelay = DHTLIB_DHT11_WAKEUP;
    rv = _read();  //  read again with correct conversion.
    return rv;
```

Getting late here.

RobTillaart, 2024-12-06:
@mitra42 

I have added the ```_wakeupDelay = DHTLIB_DHT11_WAKEUP;``` line in the .cpp
Build is now running.

If time permits could you test if the auto recognition for the KY015 works better with that extra line?

mitra42, 2024-12-06:
Will do - I've got two really busy days over the weekend, so it might not be till Monday my time

RobTillaart, 2024-12-06:
Next week is fine!
The code wont run away.

Have a good weekend!

mitra42, 2024-12-06:
Ok - tested, I can confirm that `dhtnew_simple` detects correctly - for my KY-015 it gets one "Sensor not Ready" read then starts reading correctly. 

In my own code I get two such readings - not sure what the difference is, and that is also fine..

mitra42, 2024-12-19:
Just checking - it looks like 0.5.1 is the latest release - were you intending to release 0.5.2 ?

Also - could you shoot me an email so I can send you the update on our project. Mine is mitra@Mitra.biz, I don't see yours on your git profile and I don't see any other way to message you other than in a specific issue :-)

RobTillaart, 2024-12-22:
@mitra42 

> were you intending to release 0.5.2 ?

yes, as soon as time permits, (still recovering) 
As far as I recall all was stable and working now for you.

> Also - could you shoot me an email ...

Sent one on December19.

closed 2024-12-29 by RobTillaart

RobTillaart, 2024-12-29:
@mitra42 
0.5.2 released as all looked stable,

commit 2587d004 (2024-11-26): fix #104, support KY015 (again)
commit b660430f (2024-12-29): fix #104, support KY015 (again) (#105)

- fix #104, add support for KY015 (again)
- minor edits examples
- add **DHT_endless_debug.ino** develop example